rusefi/hellen121vag #32: less crazy routing for ignition outputs

mck1117, 2021-05-20:
maybe put ignition driver module closer to output pins, instead of routing the ignition output signals half way around the board like this: 

[image]

rusefillc, 2021-05-20:
Maybe time to 4 layer?

mck1117, 2021-05-20:
I mean move the module.  The routing is fine, I'd just rather not have "outside world" signals making a huge loop area.  Much better to locate the potentially noisy drivers close to the plug, instead of clear across the board.

mck1117, 2021-06-14:
poke?

rusefillc, 2021-06-14:
We will have a bigger change with rev C while adding SD card

rusefillc, 2021-08-24:
pictured U460 is the ignition module piece

commit 835b2fd7 (2021-09-08): less crazy routing for ignition outputs fix #32
Better match injector traces fix #45
Convert to 4 layer fix #56
Connect J1-6 USB cable shield to GND fix #62
closed 2021-09-08 by rusefillc