rusefi/hellen121vag #40: Document F1/F2

rusefillc, 2021-06-08:
F1/F2 seems to be similar to R2 R3 R8 R9

I think we shall document R2/R3/R8/R9 meaning with a bit of text right on PCB?

rusefillc, 2021-06-08:
also open question of we can put any relevant text right on the PCB or not even bother?

@Qwerty-OFF maybe F1 can say "from 121" and F2 can say "from blah blah" etc? R2 could say "MR to 21"?

andreika-git, 2021-06-09:
First attempt to add F1/F2 jumpers to the H121vag wiki page:
[image]

Qwerty-OFF, 2021-06-09:
We can do something like this if we need and have enogh space on the board
[image]

rusefillc, 2021-06-11:
Looks great!

closed 2021-06-11 by rusefillc