rusefi/hellen121vag #41: TPS calibration drama

rusefillc, 2021-06-08:
is it about op-amp PNs?

Matthew Kennedy  1:27 PM
You've been struck by the woes of the LMV324

1:27
This is why we need MCP6004

Matthew Kennedy  1:35 PM
which is...unfortunate
1:35
it means ETB probably won't work without replacing those opamps

mck1117, 2021-06-08:
see https://github.com/andreika-git/hellen-one/issues/27 and https://github.com/andreika-git/hellen-one/issues/63

I ran in to exactly the same problem when using a Proteus with LMV324 on my car.  Replaced with MCP6004, problem solved.

rusefillc, 2021-06-09:
Less rail to rail comparing with 6004

mck1117, 2021-06-09:
@dron0gus LMV324 Output swing only goes from 0.05 to 4.45 volts.  MCP6004 output swing goes from 0.002 to 4.985 volts.

dron0gus, 2021-06-10:
@mck1117 thanks.
I'm using -0.6V and +5.5V to power OpAmps on my custom board. Looks like this saved me some time debuging similar issues.

closed 2021-06-11 by rusefillc