rusefi/hellen121vag #42: WBO VN1160 not acceptable

mck1117, 2021-06-08:
VN1160 doesn't work. I don't know why this chip was chosen.

Qwerty-OFF, 2021-06-09:
https://github.com/rusefi/hellen64_miataNA6_94/issues/54#issuecomment-857916625

rusefillc, 2021-06-10:
Dron_Gus  5:57 AM
@Matthew Kennedy can I replace smart mosfet in WBO with simple N-channel mosfet?
5:57
like IPD079N06 or STD25NF10

Matthew Kennedy  1:08 PM
I think it would work, but you get no protection against short circuit or anything

rusefillc, 2021-06-10:
@mck1117 can you please remove wrong part from your hellen?
@mck1117 do you have a part to install on your hellen 121?

@kon please also at least remove the mosfet to reduce confusion

mck1117, 2021-06-10:
Yes, I have a handful of leftover VNS14NV04 from another project, and already swapped one while diagnosing the problem.

BTW: that's the low side driver from Frankenso, if you have a bunch of those sitting around.

rusefillc, 2021-06-13:
@Qwerty-OFF please close once we make BOM change

closed 2021-06-13 by Qwerty-OFF