rusefi/hellen121vag #60: replace VR module with VR-discrete

rusefillc, 2021-08-24:
remove VR module
add VR-discrete like on https://github.com/rusefi/hellen128mercedes
VR_analog not needed

[image]

[image]

rusefillc, 2021-08-24:
@mck1117 what is current requirement for THRESHOLD_VR? does it have to a low side channel or could it be just an stm32 pin?

https://github.com/andreika-git/hellen-one/wiki/module-output

mck1117, 2021-08-24:
>  VR_analog not needed

Not true, this will eventually be required for configuration.

> @mck1117 what is current requirement for THRESHOLD_VR? does it have to a low side channel or could it be just an stm32 pin?

It needs a PWM output directly from the MCU, no output module.

rusefillc, 2021-08-24:
@mck1117 please elaborate
"analog" vs "configuration"
does it have to be an ADC pin or just any GPIO pin?

rusefillc, 2021-08-24:
Some notes added to https://github.com/andreika-git/hellen-one/wiki/module-vr-discrete

mck1117, 2021-08-24:
Analog in to the MCU will let you view the waveform from the sensor in real time, so you can set the threshold. Maybe completely automatic in the future.

The analog input to the MCU needs to be something on ADC3.

rusefillc, 2021-08-24:
@mck1117 is there a chance that we are over-engineering here?

mck1117, 2021-08-24:
This would be nice to do, since without the max992x we don't get automagic threshold:

https://www.maxxecu.com/webhelp/settings-diagnostics-trigger_oscilloscope.html

rusefillc, 2021-09-08:
@mck1117 can you please pick where to connect VR_analog on hellen121vag?

rusefillc, 2021-09-15:
[image]

rusefillc, 2021-09-22:
Looks like depends/related to #49

rusefillc, 2021-09-22:
Since we are removing old VR there is no need in input/output jumpers

rusefillc, 2021-09-22:
VR_THRESHOLD could use any of the GPIO unused pins

E23 E11 etc

mck1117, 2021-09-22:
> VR_THRESHOLD could use any of the GPIO unused pins
> 
> E23 E11 etc

No, VR_THRESHOLD needs a hardware PWM pin from the STM32.  It's in theory possible with soft PWM, but faster hardware PWM does make a big difference there.

rusefillc, 2021-09-22:
ok, on this board we can move idle

closed 2021-10-01 by rusefillc