rusefi/hellen121vag #62: Connect J1-6 USB cable shield to GND

rusefillc, 2021-09-07:
J1-6 USB cable shield was not connected to GND due to tracing issues with JLC tolerances.

We should connect J1-6 USB cable shield to GND once we migrate board to 4 layers

mck1117, 2021-09-07:
...how was this a problem with JLC tolerance?

The recommendation is not shorting the shield to ground.  On proteus it has a 100n + 100k in parallel to ground.

mck1117, 2021-09-07:
ok looks like there are conflicting opinions in the literature about what's best - let's put an 0603 pad for now with a 0 ohm mounted

hasl58, 2021-09-07:
>>in the literature

link please show

mck1117, 2021-09-07:
https://electronics.stackexchange.com/a/4519

hasl58, 2021-09-07:
if USB 2.0 and less pin6 to GND
if USB 3.0 and large pin6 from resistor

hasl58, 2021-09-07:
ok. i connect through resistor.

rusefillc, 2021-09-07:
@hasl58 what exact lib file would need to be changed? is that some standard library or rusEFI custom library https://github.com/rusefi/kicad-libraries/ ?

if it's https://github.com/rusefi/kicad-libraries/ we would need a pull request into https://github.com/rusefi/kicad-libraries/ - i.e. fork https://github.com/rusefi/kicad-libraries/ and etc same process as PR into this hellen121vag here

commit 835b2fd7 (2021-09-08): less crazy routing for ignition outputs fix #32
Better match injector traces fix #45
Convert to 4 layer fix #56
Connect J1-6 USB cable shield to GND fix #62
closed 2021-09-08 by rusefillc