rusefi/hellen121vag #84: 3.2 6 cylinder CAM/CRANK

rusefillc, 2021-12-17:
https://github.com/rusefi/rusefi_documentation/blob/master/OEM-Docs/VAG/2004-cayenne-3.2-all.pdf

Does this engine have four camshafts?

Camshaft sensors 1 and 2. Are those VR or Hall? Which of the four cams are those?

[image]

rusefillc, 2021-12-17:
I really hope that diagram is fucked up

at least one typo here - note x1 of "camshaft sensor 1" and x3 of "camshaft sensor 2"

but also two of "engine speed" which does not match sensors

[image]

NMSTEC, 2021-12-17:
Discharge wave, intake wave is CAM, and is Hall. 
Camshaft sensor 1 & 2 is solenoid.

It is indeed wrong Labelling. They use google translate from German. Please see my additions in Excel, as I've accounted for the errors.

NMSTEC, 2021-12-17:
Also this is not 3.2 V6. This is 3.2 VR6. Only one bank.

rusefillc, 2021-12-17:
@NMSTEC for now please wire crank into pin 54 which is VSS in TS at the moment

NMSTEC, 2021-12-17:
[3.2 vr6 touareg.pdf](https://github.com/rusefi/hellen121vag/files/7737158/3.2.vr6.touareg.pdf)
Slightly less shit diagram