rusefi/hellen121vag #90: update schematics to mega

rusefillc, 2023-03-30:


Nadezhda-n, 2023-03-31:
Also _vr-discrete_ and _wbo_?

rusefillc, 2023-03-31:
> Also **vr-discrete** and **wbo**?

Yes please!

Nadezhda-n, 2023-04-03:
Это ревизия D?

rusefillc, 2023-04-03:
> Это ревизия D?

Yep looks like we are working on rev-D at the moment

Nadezhda-n, 2023-04-05:
Я прошу проверить, верно ли я сделала вот это:
[image]
[image]

rusefillc, 2023-04-05:
SD&accel точно верно, синий блок давай сотрём
# unused очень мутное слово, давай уточним про что это всё-таки и почему

если про не-монтаж второй кан шины, то бесопаснее и мудрее не монтировать только главный чип за доллар, а рассыпуху проще монтировать

Nadezhda-n, 2023-04-05:
> SD&accel точно верно, синий блок давай сотрём
> 
> # unused
> очень мутное слово, давай уточним про что это всё-таки и почему
> 
> если про не-монтаж второй кан шины, то бесопаснее и мудрее не монтировать только главный чип за доллар, а рассыпуху проще монтировать

Именно про это и спрашивала, так и записала 
[image]

Nadezhda-n, 2023-04-05:
> Именно про это и спрашивала, так и записала [image]

убрала все изменения в этом файле